internet-innovation/nm-exp-active-netrics #2: ookla speedtest crash

ggmartins, 2021-02-26:
**Description**: Sometimes speedtest just crashes.
```
Retrieving speedtest.net configuration...
Testing from Comcast Cable (73.36.170.72)...
Retrieving speedtest.net server list...
Selecting best server based on ping...
Hosted by Comcast (Chicago, IL) [8.08 km]: 25.077 ms
Testing download speed................................................................................
Download: 170.67 Mbit/s
Testing upload speed............................................................Exception in thread Thread-71:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3/dist-packages/speedtest.py", line 905, in run
    f = self._opener(request)
  File "/usr/lib/python3.8/urllib/request.py", line 525, in open
    response = self._open(req, data)
  File "/usr/lib/python3.8/urllib/request.py", line 542, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "/usr/lib/python3.8/urllib/request.py", line 502, in _call_chain
    result = func(*args)
  File "/usr/lib/python3/dist-packages/speedtest.py", line 537, in http_open
    return self.do_open(
  File "/usr/lib/python3.8/urllib/request.py", line 1354, in do_open
    r = h.getresponse()
  File "/usr/lib/python3.8/http/client.py", line 1347, in getresponse
    response.begin()
  File "/usr/lib/python3.8/http/client.py", line 307, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.8/http/client.py", line 289, in _read_status
    raise BadStatusLine(line)
http.client.BadStatusLine: ûD]¾à]ÁSq~=äÍ3v4ÒHTTP/1.1 200 OK

..........................................
Traceback (most recent call last):
  File "/usr/bin/speedtest", line 11, in <module>
    load_entry_point('speedtest-cli==2.1.2', 'console_scripts', 'speedtest')()
  File "/usr/lib/python3/dist-packages/speedtest.py", line 1986, in main
    shell()
  File "/usr/lib/python3/dist-packages/speedtest.py", line 1951, in shell
    speedtest.upload(
  File "/usr/lib/python3/dist-packages/speedtest.py", line 1664, in upload
    self.results.bytes_sent = sum(finished)
TypeError: unsupported operand type(s) for +: 'int' and 'NoneType'
```

**Expected**:
We should handle these crashes in Popen execution and log ERROR accordingly.

tarunmangla, 2021-03-02:
Kyle is going to use the speedtest terminal client instead of the python package. The terminal client does not use HTTP, so the error handling would be different.

kyle-macmillan, 2021-03-11:
Just created a new branch with the speedtest fixes. What is the best way to test that it works? Do I need to remake the package?

ggmartins, 2021-03-15:
if the fix is simple, please feel free to merge into the main branch, happy to help with the testing.

kyle-macmillan, 2021-03-15:
sounds good, just merged

JamesSaxon, 2021-04-09:
Hi @kyle-macmillan and @ggmartins -- upping here -- 

I just noticed that the speedtest version is running `speedtest --json` rather than `speedtest -f json`.  The former is the format for the speedtest-cli version, which is having issues today, and is not officially supported.  The latter is from Ookla.  

https://github.com/chicago-cdac/nm-exp-active-netrics/blob/b11feffa13a1e369b736fd34df3bd931d5b04557/src/netrics/netson.py#L105

So I suspect we're still using the sivel version -- 

https://github.com/sivel/speedtest-cli

I confirm that here:

https://github.com/chicago-cdac/nm-exp-active-netrics/blob/b11feffa13a1e369b736fd34df3bd931d5b04557/requirements.txt#L14

We should switch to this one: 

https://www.speedtest.net/apps/cli

@kyle-macmillan, do you have time to switch, or @ggmartins could you?  I am trying to get a talk written for next week.

Thanks -- 

Jamie

commit 10dd189d (2021-04-15): fix: #2 ookla, #9 ndt7
closed 2021-04-15 by ggmartins