internet-innovation/nm-exp-active-netrics #3: better test scheduling

ggmartins, 2021-03-02:
ideal solution imho: test execution lock acquire / wait to prevent multiple executions at the same time.  

The **--get-times** generates a random time (minute) from a fixed 6-minute range calculated from taking **netrics.json** as initial "feed". 
https://github.com/chicago-cdac/nm-exp-active-netrics/blob/main/src/nmexpactive/experiment.py#L144

Maybe there's a better / more configurable way of doing this, considering the fact that we moved netrics.json to /docs for easy access/editing. Maybe the numbers in `.. + random.randrange(-3,3)` should come from netrics.json.

If this is "too much" to cover right now, maybe we can start by fixing this `1 * * * ..`, it looks like the south shore speedtest server can get hammered at that minute.
`echo "1 * * * * root /usr/local/bin/netrics -s ${upload} >>${log} 2>&1" >> /etc/cron.d/cron-nm-exp-active-netrics`
https://github.com/chicago-cdac/nm-exp-active-netrics/blob/main/etc/init.d/nm-exp-active-netrics#L39

JamesSaxon, 2021-03-03:
My suggestion would be that we do this deterministically.  Random tests will eventually overlap.  Better to just do "install 1 gets minute 1," install 2 gets minute 2."  The test lock generates a bunch of overhead, especially, since we'll have 100 installs trying to get the lock.  How does the crontab get created?  Why not just put it in the configuration?

ggmartins, 2021-03-03:
By _lock_ I meant locking local (same device) test execution, to avoid collisions between dns, udp throughput, etc; not a test server lock. For that, I don't think it's practical. Also, I don't think it's practical to generate a special config for every single install we get, I'd recommend simple randomization within the hour.

crontab generation:
https://github.com/chicago-cdac/nm-exp-active-netrics/blob/main/etc/init.d/nm-exp-active-netrics#L39

randomization concept:
https://github.com/chicago-cdac/nm-exp-active-netrics/blob/main/src/nmexpactive/experiment.py#L144

JamesSaxon, 2021-03-03:
I see.  For the local interference, I was just stickingiperf "off of the 5's" that is not on one of the */5 that the other tests take (all complete in less than a minute).

The crontab generation you linked looks like it already does the IPERF_TIME thing that I was suggesting -- so we could do that just fine for SPEEDTEST (confused).  And since we'll have to change at least the last name of each install, I think changing the time is not such a big burden.

I do think that these need to be deterministic.  With 100 devices going once every 2 hours (or more, if we add TCP), random is just not an efficient way of avoiding collisions, since these tests take a solid fraction of a minute.

Some installs may change their set of tests, so I think we have to support this.  I don't see how putting a sequential counter into configurations through the CMS (1 .. N users), to specify the minute would be such a burden.  The whole point of the CMS is to handle that, no?

JamesSaxon, 2021-05-07:
Also, what is the situation that you are "desperate" to fix -- does it just crunch out your connection experience every 2 hours?

ggmartins, 2021-05-07:
that's right, every hour. Cool, I prefer ndt for frequent measurements, so I'll do this for mine, and leave ookla as frequent in the default, thanks, G.