internet-innovation/nm-exp-active-netrics #10: Add local ndt7

JamesSaxon, 2021-04-09:
@ggmartins, @kyle-macmillan, @tarunmangla, @feamster for discussion:

We should add the ndt7 docker container, to allow for wifi tests.

On jetson, adapting the instructions and README.
* https://www.measurementlab.net/blog/run-your-own-ndt-server/#setup-and-run-an-ndt-server-on-ubuntu-1804-lts
* https://github.com/m-lab/ndt-server

```
git clone https://github.com/m-lab/ndt-server.git
cd ndt-server
docker build . -t ndt-server

# Start up docker for the jetson:
sudo apt install docker.io
sudo usermod -aG docker $USER
install -d certs datadir
./gen_local_test_certs.bash

# -d for daemon; turn off for testing.
docker run -d --network=host                            \
           --volume `pwd`/certs:/certs:ro         \
           --volume `pwd`/datadir:/datadir        \
           --read-only                            \
           --user `id -u`:`id -g`                 \
           --cap-drop=all                         \
           ndt-server                                 \
           -cert /certs/cert.pem                  \
           -key /certs/key.pem                    \
           -datadir /datadir                      \
           -ndt7_addr 192.168.1.4:33007             \
           -ndt7_addr_cleartext 192.168.1.4:38080
 ```

Then we need a little bit of javascript, adapted from here:
* https://github.com/m-lab/ndt-server/blob/master/html/ndt7.html

Note that ndt7 server side drops the json.gz in the datadir, so we could also just parse and ship, if there are any

We _do_ still probably want to allow user input, i.e., "my connection is bad," which will require something dynamic (django/flask) beyond the ndt server.

JamesSaxon, 2021-05-20:
@ggmartins , if you can advise about this, I'm happy to spend some time on it.

I don't know how setting up a docker container works WRT the CMS and existing packages.

After that, one can either:
* have netrics just grep for the files from the datadir or 
* report the result directly from the web integration

We should also add a browser extension that runs this in the background every six hours or something.

ggmartins, 2021-05-27:
We just need to pack all of these commands in one single script that, after its execution, we're 100% sure that the container is up and running. It would be easier if we could just pull the image from a public repo as opposed to building it locally for every single device. Also, if you can provide a good solution to replace that hard-coded IP address "192.168.1.4", that ideal.

closed 2021-06-15 by jesteria

jesteria, 2021-06-15:
For information about the work that resolved this issue, and how to make use of it, see [Netrics Local Dashboard Release 0.0.1](https://github.com/chicago-cdac/netrics-dash/releases/tag/0.0.1).

Note: The above makes the Local Dashboard and its basic feature set available for evaluation during R2. Further functionality such as browser-backgrounded speed tests (via browser extension) remain to be completed.