internet-innovation/nm-exp-active-netrics #9: Add NDT7 Speedtests

JamesSaxon, 2021-04-09:
We should add the ndt7 tests.

ndt7-client requires go >= 1.12; Ubuntu 18.04 is 1.10 by default, 
so check [available versions](https://golang.org/dl/).
Then per go instructions [here](https://golang.org/doc/install), download install.
```
wget https://golang.org/dl/go1.16.3.linux-arm64.tar.gz
tar -C /usr/local -xzf go1.16.3.linux-arm64.tar.gz
```

Then install the ndt7-client per [these instructions](https://github.com/m-lab/ndt7-client-go#installing)
```
git clone https://github.com/m-lab/ndt7-client-go.git
export GO111MODULE=on
cd ndt7-client-go/
go get ./cmd/ndt7-client
```

In my bashrc, I put 
```
export GOPATH=~/.bin/go/
export PATH=$PATH:/usr/local/go/bin:~/.bin/go/bin
```
Then the test is just
```
ndt7-client -format json -quiet
```

The output is json, so this should be "trivial" to add in the python .

I don't have time to do this for a week or so, since I'm not as confident with the build system.  @ggmartins, if you could incorporate the ndt7-client, I'm happy to add the python.

ggmartins, 2021-04-09:
I can take that one. Same scheduling as speedtest, just a different time? we want the tcp retransmission parsed as well, right?

JamesSaxon, 2021-04-09:
Yes, definitely grab the TCP.  I actually just tested ndt7 on tigerteam, and that is actually working OK despite the kernel vesion.  We should just use their infrastructure, I think.

closed 2021-04-16 by ggmartins