internet-innovation/nm-exp-active-netrics #14: latency under load iperf error

ggmartins, 2021-04-27:
```

 --- google ping latency under load ---
Packet Loss Under Load: 0.0%
Average RTT Under Load: 187.713 (ms)
Minimum RTT Under Load: 111.525 (ms)
Maximum RTT Under Load: 256.142 (ms)
RTT Std Dev Under Load: 42.515 (ms)
iperf3: error - unable to send control message: Connection reset by peer

 --- google ping latency under load ---
Packet Loss Under Load: 0.0%
Average RTT Under Load: 13.847 (ms)
Minimum RTT Under Load: 11.297 (ms)
Maximum RTT Under Load: 20.74 (ms)
RTT Std Dev Under Load: 2.701 (ms)
{'speedtest_ookla_download': 578.77356, 'speedtest_ookla_upload': 16.84809, 'speedtest_ookla_jitter': 8.005, 'speedtest_ookla_latency': 15.397, 'speedtest_ookla_pktloss': 0, 'speedtest_ndt7_download': 425.99939920796345, 'speedtest_ndt7_upload': 19.084749348277708, 'speedtest_ndt7_downloadretrans': 0.0, 'speedtest_ndt7_minrtt': 6.937, 'google_packet_loss_pct_under_ul': 0.0, 'google_rtt_min_ms_under_ul': 111.525, 'google_rtt_max_ms_under_ul': 256.142, 'google_rtt_avg_ms_under_ul': 187.713, 'google_rtt_mdev_ms_under_ul': 42.515, 'google_packet_loss_pct_under_dl': 0.0, 'google_rtt_min_ms_under_dl': 11.297, 'google_rtt_max_ms_under_dl': 20.74, 'google_rtt_avg_ms_under_dl': 13.847, 'google_rtt_mdev_ms_under_dl': 2.701}
```

ggmartins, 2021-04-27:
--- NETWORK MEASUREMENTS ---
iperf3: error - the server is busy running a test. try again later

 --- google ping latency under load ---
Packet Loss Under Load: 0.0%
Average RTT Under Load: 14.425 (ms)
Minimum RTT Under Load: 11.389 (ms)
Maximum RTT Under Load: 20.451 (ms)
RTT Std Dev Under Load: 2.54 (ms)
iperf3: error - the server is busy running a test. try again later

@JamesSaxon how can we fix this?

JamesSaxon, 2021-04-28:
Hey, I saw this one too.  My "gut" here is to just parse the message and drop iperf on "busy", so it fails gracefully -- i.e., if we have other speedtests or pings, we don't want to lose those.

The usual problem is just that two people are using the system at once.  Strictly speaking, that shouldn't happen (if we manage the cron jobs right), but it is bound to, in testing.

We could also open more ports on the server, so long as we are **very** confident that we have the capacity.  But obviously, that doesn't work if we are testing from your or Nick's ~gigabit connections.

ggmartins, 2021-05-26:
This will be addressed primarily through a "precise" allocation of the measurement servers for each of the devices as suggested by @JamesSaxon.  As such, this becomes a task for cms.

ggmartins, 2021-07-14:
This is being replaced by the new LuL tool.

closed 2021-07-14 by ggmartins