internet-innovation/nm-exp-active-netrics #33: speedtest netson code JSONDecodeError

ggmartins, 2021-05-19:
``` 
 ==============================================================================
    
    You may only use this Speedtest software and information generated
    from it for personal, non-commercial use, through a command line
    interface on a personal computer. Your use of this software is subject
    to the End User License Agreement, Terms of Use and Privacy Policy at
    these URLs:
    
    	https://www.speedtest.net/about/eula
    	https://www.speedtest.net/about/terms
    	https://www.speedtest.net/about/privacy
    
    ==============================================================================
    
    License acceptance recorded. Continuing.
    
    
     --- NETWORK MEASUREMENTS ---
    
     --- Ookla speed tests ---
    Download:	476.695408 Mb/s
    Upload:		23.909872 Mb/s
    Latency:	14.055 ms
    Jitter:		2.48 ms
    PktLoss:	0 Total Count
    Traceback (most recent call last):
      File "./src/netrics.py", line 245, in <module>
        output['ookla'], output['ndt7'] = test.speed()
      File "/usr/local/src/nm-exp-active-netrics/src/netrics/netson.py", line 165, in speed
        return self.speed_ookla(True), self.speed_ndt7(True)
      File "/usr/local/src/nm-exp-active-netrics/src/netrics/netson.py", line 143, in speed_ndt7
        res_json = json.loads(output)
      File "/usr/lib/python3.8/json/__init__.py", line 357, in loads
        return _default_decoder.decode(s)
      File "/usr/lib/python3.8/json/decoder.py", line 340, in decode
        raise JSONDecodeError("Extra data", s, end)
    json.decoder.JSONDecodeError: Extra data: line 2 column 1 (char 81)
```

kyle-macmillan, 2021-06-02:
Is this error still getting thrown? If so I can take a look.

ggmartins, 2021-06-02:
I saw this just once. I've just scanned the logs and self-checked output, no signs of it.

closed 2021-07-14 by ggmartins