internet-innovation/nm-exp-active-netrics #48: hops-to-backbone measurement error on ATT

ggmartins, 2021-06-12:
I think what you're trying to do here is specific to Comcast and probably won't work with other ISPs.
https://github.com/chicago-cdac/nm-exp-active-netrics/blob/main/src/netrics/netson.py#L324
```
traceroute -m 15 -N 32 -w3 www.google.com | grep -m 1 ibone
 6  be-32221-cs02.350ecermak.il.ibone.comcast.net (96.110.40.53)  17.727 ms be-32231-cs03.350ecermak.il.ibone.comcast.net (96.110.40.57)  17.889 ms  17.135 ms
```
Here's our ATT's output:
```
    Traceback (most recent call last):
      File "./src/netrics.py", line 269, in <module>
        output['hops_to_backbone'] = test.hops_to_backbone(args.backbone)
      File "/usr/local/src/nm-exp-active-netrics/src/netrics/netson.py", line 331, in hops_to_backbone
        hops = int(tr_res_s[0])
    ValueError: invalid literal for int() with base 10: ''
```

kyle-macmillan, 2021-06-13:
I think the intention was to understand how many hops it took to get out of the city (debatable how useful this is) by counting the number of hops to what i think is an IXP that most of the comcast traffic travels through. I think its still worth keeping since Comcast is still the dominant provider so I'll add an exception to this test that submits -1 if the above error is thrown.

JamesSaxon, 2021-06-13:
Yes -- agree with Kyle's description.  If we could somehow traceroute "350
E Cermak," that would be ideal.  Or figure out the appropriate "ibone"
replacement to grep for, with AT&T.

The ideal would be hops to some resources that we *know* is geographically
situated on the East or West Coasts **AND** that generates a valid,
comprehensible, and immediate traceroute.  For example, cs.berkeley.edu and
wikipedia.org just "star-ing out" / no valid response post the backbone,
even though a ping is valid.

ggmartins, 2021-06-14:
I'd ask advice from CAIDA (Matt Lucke, for example) before attempting anything too complex there, I want to make sure we're not reinventing the wheel. We have some experience incorporating his scamper tool in the past, so maybe we can get some insights on that too (BTW, I see us adding that tool soon or later).

ggmartins, 2021-06-18:
I'm closing this for now for the lack of better solutions. Please keep in mind we'll have some level of warning counted in our logs:
```
    log_lines: 77
    log_lines_cron: 154
    log_lines_warns: 3
    log_lines_error: 0

    2021-06-18T17:27:37.798 WARNING {netson} [hops_to_backbone] hops_to_backbone invalid literal for int() with base 10: ''
```

closed 2021-06-18 by ggmartins