internet-innovation/nm-exp-active-netrics #71: oplat runtime error: index out of range

ggmartins, 2021-09-30:
```
ERROR in /var/nm/nm-exp-active-netrics/log/nm-exp-active-netrics.log: b'2021-09-30T00:06:22.178 ERROR {netson} [ping_latency] ping: www.youtube.com: Temporary failure in name resolution\n')
ERROR in /var/nm/nm-exp-active-netrics/log/nm-exp-active-netrics.log: b'2021-09-30T00:06:42.218 ERROR {netson} [ping_latency] ping: www.facebook.com: Temporary failure in name resolution\n')
ERROR in /var/nm/nm-exp-active-netrics/log/nm-exp-active-netrics.log: b'2021-09-30T03:30:29.556 ERROR {netson} [ping_latency] ping: www.facebook.com: Temporary failure in name resolution\n')
ERROR in /var/nm/nm-exp-active-netrics/log/nm-exp-active-netrics.log: b'2021-09-30T12:17:13.542 ERROR {netson} [speed_ookla] {"type":"log","timestamp":"2021-09-30T12:17:13Z","message":"Failed to resolve host name. Cancelling test suite.","level":"error"}\n')
ERROR in /var/nm/nm-exp-active-netrics/log/nm-exp-active-netrics.log: b'2021-09-30T12:18:02.571 ERROR {netson} [oplat] panic: runtime error: index out of range [-1]\n')
ERROR in /var/nm/nm-exp-active-netrics/log/nm-exp-active-netrics.log: b'2021-09-30T15:18:26.503 ERROR {netson} [oplat] 2021/09/30 15:18:26 iPerf3 finished before pings, suggested test length increase: 35.708946ms\n')
ERROR in /var/nm/nm-exp-active-netrics/log/nm-exp-active-netrics.log: b'2021-09-30T15:18:46.654 ERROR {netson} [oplat] panic: runtime error: index out of range [-1]\n')
ERROR in /var/nm/nm-exp-active-netrics/log/nm-exp-active-netrics.log: b'2021-09-30T18:10:30.608 ERROR {netson} [ping_latency] ping: www.facebook.com: Temporary failure in name resolution\n')
```

kyle-macmillan, 2021-09-30:
It looks like oplat exited because iperf3 finished before it heard from the pings. I can have it fail gracefully. What value are we using to indicate that the test failed?

JamesSaxon, 2021-09-30:
Just heads up that I think this is a connectivity failure that is not
completely protected, not just a slow response -- `Temporary failure in
name resolution` suggests offline.

So this looks like a bad connection, but yes we have to fail gracefully.
Note that Speedtest also crashed out, not just oplat.

Right now, tests proceed if any of Google, Amazon, or Wikipedia succeed:

https://github.com/chicago-cdac/nm-exp-active-netrics/blob/943163a9f52d02dc973fe31918c0ba7f6dbf84df/src/netrics.py#L322

Any other suggestions for conditions to stop tests?

Was this on ... NW's, by any chance?

ggmartins, 2021-09-30:
yes, NW.

kyle-macmillan, 2021-09-30:
I have updated oplat to fail gracefully when it does not hear back from any probes (the cause of the runtime error).

closed 2021-09-30 by kyle-macmillan

JamesSaxon, 2021-09-30:
How did you test?