internet-innovation/nm-exp-active-netrics #81: Malformed JSON

marcwitasee, 2022-02-07:
Some JSONs are malformed:

`{"Measurements":{"connected_devices_arp":{"devices_1day":8,"devices_1week":8,"devices_active":4,"devices_total":23},"dns_latency":{"dns_query_avg_ms":859.4,"dns_query_max_ms":3907,"error":false},"hops_to_target":{"error":false,"hops_to_google":12},"ipquery":{"ipv4":"174.62.179.118"},"last_mile_rtt":{"Cloudflare_DNS_last_mile_ping_packet_loss_pct":9.09091,"Cloudflare_DNS_last_mile_ping_rtt_avg_ms":257.32,"Cloudflare_DNS_last_mile_ping_rtt_max_ms":276.617,"Cloudflare_DNS_last_mile_ping_rtt_mdev_ms":12.665,"Cloudflare_DNS_last_mile_ping_rtt_min_ms":222.905,"Cloudflare_DNS_last_mile_tr_rtt_max_ms":8.945,"Cloudflare_DNS_last_mile_tr_rtt_median_ms":8.363,"Cloudflare_DNS_last_mile_tr_rtt_min_ms":8.318},"ping_latency":{"Atlanta_packet_loss_pct":9.09091,"Atlanta_rtt_avg_ms":287.162,"Atlanta_rtt_max_ms":290.533,"Atlanta_rtt_mdev_ms":1.19,"Atlanta_rtt_min_ms":286.23,"Chicago_packet_loss_pct":9.09091,"Chicago_rtt_avg_ms":264.164,"Chicago_rtt_max_ms":275.892,"Chicago_rtt_mdev_ms":5.027,"Chicago_rtt_min_ms":260.649,"Denver_packet_loss_pct":9.09091,"Denver_rtt_avg_ms":285.121,"Denver_rtt_max_ms":286.802,"Denver_rtt_mdev_ms":0.619,"Denver_rtt_min_ms":284.582,"Hong_Kong_packet_loss_pct":9.09091,"Hong_Kong_rtt_avg_ms":457.083,"Hong_Kong_rtt_max_ms":462.003,"Hong_Kong_rtt_mdev_ms":1.673,"Hong_Kong_rtt_min_ms":455.981,"Johannesburg_packet_loss_pct":9.09091,"Johannesburg_rtt_avg_ms":448.975,"Johannesburg_rtt_max_ms":475.601,"Johannesburg_rtt_mdev_ms":68.549,"Johannesburg_rtt_min_ms":243.388,"Paris_packet_loss_pct":9.09091,"Paris_rtt_avg_ms":354.864,"Paris_rtt_max_ms":357.294,"Paris_rtt_mdev_ms":1.128,"Paris_rtt_min_ms":353.079,"Sao_Paulo_packet_loss_pct":9.09091,"Sao_Paulo_rtt_avg_ms":397.115,"Sao_Paulo_rtt_max_ms":399.295,"Sao_Paulo_rtt_mdev_ms":0.98,"Sao_Paulo_rtt_min_ms":395.81,"Seattle_packet_loss_pct":9.09091,"Seattle_rtt_avg_ms":305.732,"Seattle_rtt_max_ms":319.362,"Seattle_rtt_mdev_ms":18.3,"Seattle_rtt_min_ms":251.397,"Stockholm_packet_loss_pct":9.09091,"Stockholm_rtt_avg_ms":348.875,"Stockholm_rtt_max_ms":400.414,"Stockholm_rtt_mdev_ms":75.044,"Stockholm_rtt_min_ms":125.401,"Sydney_packet_loss_pct":9.09091,"Sydney_rtt_avg_ms":470.422,"Sydney_rtt_max_ms":473.242,"Sydney_rtt_mdev_ms":1.254,"Sydney_rtt_min_ms":469.114,"Tunis_packet_loss_pct":9.09091,"Tunis_rtt_avg_ms":359.709,"Tunis_rtt_max_ms":405.63,"Tunis_rtt_mdev_ms":66.915,"Tunis_rtt_min_ms":160.396,"Washington_DC_packet_loss_pct":9.09091,"Washington_DC_rtt_avg_ms":266.797,"Washington_DC_rtt_max_ms":296.212,"Washington_DC_rtt_mdev_ms":42.267,"Washington_DC_rtt_min_ms":141.045,"amazon_packet_loss_pct":0,"amazon_rtt_avg_ms":25.63,"amazon_rtt_max_ms":41.916,"amazon_rtt_mdev_ms":6.681,"amazon_rtt_min_ms":13.896,"error":false,"facebook_packet_loss_pct":0,"facebook_rtt_avg_ms":259.211,"facebook_rtt_max_ms":266.892,"facebook_rtt_mdev_ms":4.814,"facebook_rtt_min_ms":246.458,"google_packet_loss_pct":9.09091,"google_rtt_avg_ms":215.571,"google_rtt_max_ms":291.675,"google_rtt_mdev_ms":95.057,"google_rtt_min_ms":21.293,"suntimes_packet_loss_pct":0,"suntimes_rtt_avg_ms":141.85,"suntimes_rtt_max_ms":264.938,"suntimes_rtt_mdev_ms":123.088,"suntimes_rtt_min_ms":18.762,"tribune_packet_loss_pct":9.09091,"tribune_rtt_avg_ms":194.622,"tribune_rtt_max_ms":291.103,"tribune_rtt_mdev_ms":108.463,"tribune_rtt_min_ms":13.948,"uchicago_packet_loss_pct":9.09091,"uchicago_rtt_avg_ms":237.44,"uchicago_rtt_max_ms":272.7,"uchicago_rtt_mdev_ms":69.935,"uchicago_rtt_min_ms":28.728,"wikipedia_packet_loss_pct":9.09091,"wikipedia_rtt_avg_ms":236.224,"wikipedia_rtt_max_ms":288.114,"wikipedia_rtt_mdev_ms":91.73,"wikipedia_rtt_min_ms":38.436,"youtube_packet_loss_pct":9.09091,"youtube_rtt_avg_ms":258.67,"youtube_rtt_max_ms":266.289,"youtube_rtt_mdev_ms":7.65,"youtube_rtt_min_ms":236.811},"test_bytes_consumed":0,"total_bytes_consumed":6019333667},"Meta":{"Extended":{"Annotation":null,"dataver":"1","debhash":"d74a91d0f11fdb31dff65b243afff1ce5c0d3002","gitlog":"fix: max_monthly checks"},"Id":"nm-mngd-20210924-77f105f0","Time":1643973307.88513}}": 859.4, "dns_query_max_ms": 3907, "error": false}, "hops_to_target": {"hops_to_google": 12, "error": false}, "connected_devices_arp": {"devices_active": 4, "devices_total": 23, "devices_1day": 8, "devices_1week": 8}, "test_bytes_consumed": 0}}`

jesteria, 2022-02-08:
@marcwitasee Does the example JSON indicate a malformed structure or the expected structure?

marcwitasee, 2022-02-08:
Just updated the description. It's an example of a malformed json.

jesteria, 2022-02-09:
Oh wow that is screwy. FTR: it's quite hard to tell what's wrong with such a large structure; luckily, computers: "line 148 column 1":

```json
{
  "Measurements":{
    …
  },
  "Meta":{
    …
  }
}
},
"dns_latency":{
   "dns_query_avg_ms":28.6,
   "dns_query_max_ms":67,
   "error":false
},
"hops_to_target":{
   "hops_to_google":10,
   "error":false
},
"connected_devices_arp":{
   "devices_active":21,
   "devices_total":74,
   "devices_1day":25,
   "devices_1week":33
},
"test_bytes_consumed":0
}
}
```

So it looks like a regular measurements file with a few subsections recopied at the bottom. That's a mess.

**If** the code that writes this is _just_ taking a proper Python `dict` and writing it in one swoop, _e.g._:

```python
with open(…) as fd:
    json.dump(data, fd)
```

…Then I don't really see how the code could be responsible, and rather the sync system might be doing something very bad.

(Of course the writer could be doing all sorts of things as well.)

ggmartins, 2022-07-08:
seems to be fixed

closed 2022-07-08 by ggmartins