internet-innovation/nm-exp-active-netrics #82: Empty JSON results on Tigerteam

marcwitasee, 2022-02-07:
Some empyt JSONs are being synced to tigerteam. Could be an error with the sync function. Or the results being saved.

Example:
`/scratch/datasets/broadband/dev1/nm-exp-active-netrics/data/s3/sync/netrics_mngd/nm-exp-active-netrics/default/nm-mngd-20210927-8da43959/20211124/json/nm_data_20211124_071607_netrics_results.json`

jesteria, 2022-02-08:
Worth checking that the sync'd file contents match the original on the device. That said, if the original file was moved as part of the sync process, that also could have introduced an issue. Alternatively, of course, the original may have been inappropriately written empty.

marcwitasee, 2022-02-08:
The same file on the device in question is also empty (i.e., it matches the sync'd file content).

```
root@wilson:/# ssx 3959 "cat /var/nm/nm-exp-active-netrics/upload/archive/default/json/nm_data_20211124_071607_netrics_results.json"
ENV enable: true
['3959'] negation: False
['3959']
INFO: using [salt  "nm-mngd-20210927-8da43959" cmd.run cmd="cat /var/nm/nm-exp-active-netrics/upload/archive/default/json/nm_data_20211124_071607_netrics_results.json"] as execution paramenter.
nm-mngd-20210927-8da43959:
```

jesteria, 2022-02-09:
OK. Might want to ensure that the data writer code _can't_ write empty files. But, I'll say, I believe I've seen this before, or at least discussed the possibility, of the sync process doing this. The problem is that the sync process is triggered as soon as something is written; but, I don't know that it waits for the file to be _closed_. (inotify supports watching for "close write", and presumably that's what we want; however, I'm not sure that our code does that. Rather, it might be watching merely for "create".) As such, it's possible that we have a race condition between the files being completely written and the sync process uploading them and moving them to the archive.

closed 2022-09-29 by marcwitasee