internet-innovation/nm-exp-active-netrics #85: error last_mile_latency

ggmartins, 2022-07-01:
This error is crashing netrics preventing all other tests to run.
```
--- NETWORK MEASUREMENTS ---
Traceback (most recent call last):
File "./src/netrics.py", line 334, in <module>
output['last_mile_rtt'] = test.last_mile_latency('last_mile_rtt', args.last_mile_rtt)
File "/usr/local/src/nm-exp-active-netrics/src/netrics/netson.py", line 508, in last_mile_latency
res.sort()
UnboundLocalError: local variable 'res' referenced before assignment
```

tarunmangla, 2022-07-01:
I believe this should work. 

We may eventually also want to fix the parsing logic causing this issue. @ggmartins  do you know which device is facing this error? We can look at the raw logs to figure out what is causing it.

ggmartins, 2022-07-01:
3db3

PR #86 (2022-07-07): hotfix for #85
Checking to see if `res` has been populated before accessing its members. Fixes #85
commit 1526c80d (2022-07-26): Merge pull request #88 from chicago-cdac/kyle/lml-error

fixes #85
closed 2022-07-26 by ggmartins