internet-innovation/nm-exp-active-netrics #92: Ookla Errors

marcwitasee, 2022-07-22:
We are seeing a handful of Ookla errors (separate from the JSON error mentioned in #80). A list of the known errors is below with device IDs where they have occurred.

```
2022-07-22T10:01:19.315 ERROR {netson} [speed_ookla] [2022-07-22 10:01:19.310] [error] Configuration - Couldn't connect to server (Network is unreachable) (see, e.g., nm-mngd-20210927-d8cfa28a)
2022-07-22T13:01:31.672 ERROR {netson} [speed_ookla] {"type":"log","timestamp":"2022-07-22T13:01:31Z","message":"Failed to resolve host name. Cancelling test suite.","level":"error"} (see, e.g., nm-mngd-20210927-d8cfa28a)
2022-07-22T14:01:21.652 ERROR {netson} [speed_ookla] [2022-07-22 14:01:16.130] [error] Trying to get interface information on non-initialized socket. (see, e.g., nm-mngd-20210927-d8cfa28a)
2022-07-22T09:47:39.429 ERROR {netson} [speed_ookla] {"type":"log","timestamp":"2022-07-22T09:47:08Z","message":"Error: [11] Cannot read: Resource temporarily unavailable","level":"error"} (see, e.g., nm-mngd-20220506-2f996174)
```

Ideally, we can solve these errors to prevent them from happening. At a minimum, we should make sure that we are catching these errors and making sure that they don't crash the NDT test that runs immediately after the Ookla test.

marcwitasee, 2022-08-01:
Related:

For some devices (e.g., [4998](https://wilson.cs.uchicago.edu:8088/d/QGEUnyf7k/20210915-8dd84998?orgId=1)), the Ookla tests are running very infrequently. As a result, almost no Ookla data is being collected from these devices.

ggmartins, 2022-08-08:
We depend on the stability of the speedtest cli command to run throughput test smoothly. Unfortunately the code is proprietary/closed and without a clear roadmap of bugfix/new versions. If you run the command apart from our system you likely to see these same errors.  So, there's nothing that comes to my mind that we can use to fix it but we can work together with ookla on task force to up the game for that code (we would mostly help with testing). As for 4998. That device, in particular, is running into some kind of ipv6 configuration issue.

closed 2022-09-29 by marcwitasee