internet-innovation/nm-exp-active-netrics #93: Ping Latency Errors

marcwitasee, 2022-07-22:
Multiples devices are experiencing a relatively high volume of Ping Latency errors. See the below list of errors logged on devices.

```
2022-07-22T09:00:01.951 ERROR {netson} [ping_latency] ping: www.google.com: Temporary failure in name resolution (nm-mngd-20210915-a64ec38c)
2022-07-22T00:35:17.251 ERROR {netson} [ping_latency] ping: connect: Network is unreachable (nm-mngd-20210915-612a3db3)
```

We should investigate these errors and see if there is anything we can do to resolve them.

marcwitasee, 2022-08-01:
Fix deployed to Wilson

marcwitasee, 2022-08-01:
Still seeing some odd ping latency errors (e.g., [3db3](https://wilson.cs.uchicago.edu:8088/d/UxHTqzXnz/20210915-612a3db3?orgId=1)).

Here is the output when I run `netrics -p` on 3db3:

```
root@wilson:/# ssx 3db3 "netrics -p"
ENV enable: true
['3db3'] negation: False
['3db3']
INFO: using [salt  "nm-mngd-20210915-612a3db3" cmd.run cmd="netrics -p"] as execution paramenter.
nm-mngd-20210915-612a3db3:

--- NETWORK MEASUREMENTS ---
{'total_bytes_consumed': 0, 'ipquery': {'ipv4': '69.246.236.2'}, 'ping_latency': {'google_error': 'Probe IndexError', 'amazon_error': 'Probe IndexError', 'wikipedia_error': 'Probe IndexError', 'youtube_error': 'Probe IndexError', 'facebook_error': 'Probe IndexError', 'tribune_error': 'Probe IndexError', 'suntimes_error': 'Probe IndexError', 'uchicago_error': 'Probe IndexError', 'Chicago_error': 'Probe IndexError', 'Washington_DC_error': 'Probe IndexError', 'Atlanta_error': 'Probe IndexError', 'Denver_error': 'Probe IndexError', 'Seattle_error': 'Probe IndexError', 'Hong_Kong_error': 'Probe IndexError', 'Sydney_error': 'Probe IndexError', 'Johannesburg_error': 'Probe IndexError', 'Tunis_error': 'Probe IndexError', 'Paris_error': 'Probe IndexError', 'Stockholm_error': 'Probe IndexError', 'Sao_Paulo_error': 'Probe IndexError', 'error': True}}
Consumption UPDATE: 0 of 1214193998 bytes (1.1GiB).
```

ggmartins, 2022-08-01:
Yes, this device, in particular, is unable to send/receive icmp. Not sure, if there's anything I can do about it.

closed 2022-09-29 by marcwitasee