internet-innovation/nm-exp-active-netrics #94: Encrypted DNS response time measurement

marcwitasee, 2022-07-26:


marcwitasee, 2022-07-26:
Modify the logic of the `dns_latency` function in netson.py to include an option to run an encrypted DNS response time test for a given set of resolvers and a given set of domains.

To implement this measurement, we will need to update the version (need to determine which version, possibly 9.17) of DiG that is running on our devices that supports DNS encryption.

feamster, 2022-08-04:
Austin also indicated to us that the latest versions of dig now support +https.  It is newer than the one with the distribution of debian we have on some of our machines so I expect the RPis may need some update.

marcwitasee, 2022-08-04:
Need to update the `dns_latency` function to run encrypted DNS lookup

- Encrypted DNS lookup DiG command `dig +https @doh.example.com isc.org A`

Need to update toml configuration file

Need to understand the types of errors that are thrown when running this specific type of DiG command

marcwitasee, 2022-08-04:
Here is the release announcement for DiG that supports DNS-over-HTTPS https://www.isc.org/blogs/bind-doh-update-2021/

marcwitasee, 2022-08-05:
@ranyasharma @feamster @marcwitasee @ggmartins 
Hey there,

you should have access to that device by now. You must be connected to tigerteam first.

```
ssh feaster@tigerteam.io
ssh ranyasharma@tigerteam.io
```
(please use the pub key you sent me)

After that, you can jump to a shared account we have at the development device using

`ssh ubuntu@192.168.111.1`

while setting this up, I realized we could have an isolated environment / separate accounts for each one of us. So, I'm working on this now with Marc. But for some reason we still want to share this account, here are some instructions:

- we need to clone git@github.com:chicago-cdac/nm-exp-active-netrics.git or git@github.com:ranyasharma/nm-exp-active-netrics.git, but for that the pub key of the shared account (ubuntu) needs to be added to your github trusted ssh keys https://github.com/settings/keys
- there should be a copy of your .gitconfig with a name like .gitconfig-ranya. The trick we do lies on the ability to switch between configs depending on the dir we are at original .gitconfig:

```
[includeIf "gitdir:~/marc/"]
        path = ~/.gitconfig-marc
[includeIf "gitdir:~/gmartins/"]
        path = ~/.gitconfig-martins
[includeIf "gitdir:~/feamster/"]
        path = ~/.gitconfig-feamster
[includeIf "gitdir:~/ranya/"]
        path = ~/.gitconfig-ranya
```

​Now you can move to the directory eg. `cd ~/ranya` & `git clone git@github.com:chicago-cdac/nm-exp-active-netrics.git`

Next, you are ready to start modifying the code, deploying and testing following what's in the `make​ help` instructions

hope this helps, please give me a few hours to set up a separate account, I think we should have that option.

I'll be available for any questions, thanks,

G

marcwitasee, 2022-08-05:
Per Guilherme:

@ranyasharma @feamster 

ok, I've set up separate accounts for us on that netrics device. Now, from tigerteam, to access the device, you can:

```
ssh mr@192.168.111.1
ssh ranyasharma@192.168.111.1
ssh feamster@192.168.111.1
```

After that, you need to generate a ssh pair (run `ssh-keygen`) and upload the pub key to your trusted keys to your GitHub account in order to clone netrics. Once you clone the repo the process remain the same, but we need to coordinate among us and see who is going to be let the code testing etc. (right now, I don't have anything in the roadmap other than helping you)

Thanks,

G

marcwitasee, 2022-08-05:
@ranyasharma @feamster @ggmartins 

Steps to follow after successfully cloning `nm-exp-active-netrics` repo into your home directory to be able to run `./netrics` for one-off testing from the cloned repository

1. Copy `go` binary into the home directory
```
sudo cp ~ubuntu/go1.17/ ~/ -R
ln -s go1.17/go go
```
2. Add path to go binary to your $PATH variable. Add the line `export PATH=$PATH:~/go/bin/`and log out and log back into your user on the netrics device. Run `echo $PATH` after logging back in to make sure that the go path is added.
```
cd
echo "export PATH=$PATH:~/go/bin/" >> .bashrc
```
3. Change to the cloned `nm-exp-active-netrics` directory and run `make ndt`
4. From the cloned directory, run `make oplat`
5. From the cloned directory, run `sudo make iperf`
6. From the cloned directory, run `make speedtest`
7. From the cloned directory, run `sudo make deps`
8. From the cloned directory, create a virtual environment by running:
```
python3 -m venv venv
. ./venv/bin/activate
python3 -m pip install -r ./requirements.txt
```
9. Deactivate environment with `deactivate`
10. Copy the local install TOML file to the `conf` directory in the clone repository
```
sudo cp /etc/nm-exp-active-netrics/nm-exp-active-netrics.toml ./conf/
```
11. Run `./netrics -k` to test that the local Netrics binary works

marcwitasee, 2022-08-18:
# Upgrading DIG

The current version of `dig` installed on our devices does not support encrypted DNS lookups.

Current version installed: DiG 9.16.1-Ubuntu
Needed version: 9.17 or higher

## Local install of updated DiG

We have a local version of `dig` (version 9.18.5) installed in the directory `/usr/local/dig/`. If you want to run dig with the updated version, be sure to specify the path to the updated binary (`/usr/local/dig/bin/dig`).

marcwitasee, 2022-08-31:
@ranyasharma 

Plan for development:

Git:

> Create a new branch for your changes: `git checkout -b encrypted-dns`
>
> Be sure to ONLY commit the changes that you are making to the `netson.py` and `netrics.py` files.

In `netson.py`:

> Copy the `dns_latency` function and rename it `encrypted_dns_latency`.
>
> Write the function logic as needed to run your encrypted DNS measurements.

In `netrics.py`:

> Add a new flag to the arguments
>
> When the flag is included in a call to `netrics` binary, add logic to call your `encrypted_dns_latency` function. (Similar to what you were doing when you were originally testing the encrypted DNS measurements in your forked repository).

To test your new function, do the following:

> Copy the global TOML file to your development repository. From your repository, run `sudo cp /etc/nm-exp-active-netrics/nm-exp-active-netrics.toml conf/`
>
> Edit the TOML file in `conf/` to change the `topic` field from "default" to "testing"
>
> Turn off the collect package while you are testing: `/etc/init.d/nm-mgmt-collectd-http stop`. This is necessary so that you don't send the test results to the backend infrastructure.
>
> Test your function by running (from your repository) `./netrics -[your-new-flag]`

ranyasharma, 2022-09-06:
Updates since our conversation on 8/31:

- I added the resolvers to the conf file (https://github.com/chicago-cdac/nm-exp-active-netrics/blob/encrypted-dns/conf/nm-exp-active-netrics.toml#L68)
- I created the encrypted_dns_latency function in the netson.py file and added the logic with the nested for loop (one for sites and one for targets) (https://github.com/chicago-cdac/nm-exp-active-netrics/blob/encrypted-dns/src/netrics/netson.py#L764)
- In netrics.py, I added a new flag to the arguments called -e  (https://github.com/chicago-cdac/nm-exp-active-netrics/blob/encrypted-dns/src/netrics.py#L200) and I also added the logic to call the encrypted_dns_latency function (https://github.com/chicago-cdac/nm-exp-active-netrics/blob/encrypted-dns/src/netrics.py#L357)

ranyasharma, 2022-09-14:
@feamster Can you please review the code for the function that runs the encrypted DNS measurements? It is located in the [netson.py](https://github.com/chicago-cdac/nm-exp-active-netrics/blob/encrypted-dns/src/netrics/netson.py) file and is called [encrypted_dns_latency](https://github.com/chicago-cdac/nm-exp-active-netrics/blob/encrypted-dns/src/netrics/netson.py#L765)

feamster, 2022-09-14:
Yes, will take a look!

ranyasharma, 2022-10-16:
Updates: 
- Created make_dig.sh script so that the version of dig that we are currently using is installed on other devices. 
- Currently testing the make_dig.sh in a Docker container

marcwitasee, 2023-03-23:
@ranyasharma and @ggmartins Was there a recent deployment of this test? Can one of you please provide an update here? Thanks!

ggmartins, 2023-03-23:
yes, we deployed it on iotlab's device. There are things to fix before going to a larger deployment