internet-innovation/nm-exp-active-netrics #98: which sites are we performing DNS lookups to?

feamster, 2022-08-04:
https://github.com/chicago-cdac/nm-exp-active-netrics/blob/1526c80d30d55cd5437032c2767370bff5f4db7c/src/netrics/netson.py#L714

From my read of this code, it looks like we're pulling in all sites from reference_site_dict, which includes IP addresses. That would yield a bunch of NXDOMAIN responses and may be affecting our query response times in a funny way.

Is this what's going on? If so we should fix this, as we are going to be dropping in a new site and target list for encrypted DNS anyhow.

marcwitasee, 2022-08-05:
@feamster @ggmartins 

Confirming that the current [`dns_latency` function in `netson.py`](https://github.com/chicago-cdac/nm-exp-active-netrics/blob/main/src/netrics/netson.py#L696) is running DNS lookups on IP addresses in the [`reference_site_dict` variable in the TOML file](https://github.com/chicago-cdac/nm-exp-active-netrics/blob/main/conf/nm-exp-active-netrics.toml#L32-L52). See output below from `netrics -d`

```
mr@netrics:~/nm-exp-active-netrics$ ./netrics -d

 --- NETWORK MEASUREMENTS ---

 --- google ping latency (MANDATORY) ---
Packet Loss: 0.0%
Average RTT: 1.976 (ms)
Minimum RTT: 2.164 (ms)
Maximum RTT: 2.611 (ms)
RTT Std Dev: 0.16 (ms)
www.google.com
www.amazon.com
www.wikipedia.org
www.youtube.com
www.facebook.com
www.chicagotribune.com
chicago.suntimes.com
cs.uchicago.edu
4.71.251.129
4.35.238.193
4.71.254.129
4.34.58.1
38.102.0.65
183.178.65.1
175.45.79.1
196.24.45.129
41.231.21.1
77.67.119.129
195.89.146.193
190.98.158.1

 --- DNS Delays (n = 20) ---
Avg DNS Query Time: 19.6 ms
Max DNS Query Time: 79 ms
{'total_bytes_consumed': 0, 'ipquery': {'ipv4': '128.135.11.14'}, 'ping_latency': {'google_packet_loss_pct': 0.0, 'google_rtt_min_ms': 1.976, 'google_rtt_max_ms': 2.611, 'google_rtt_avg_ms': 2.164, 'google_rtt_mdev_ms': 0.16, 'error': False}, 'dns_latency': {'dns_query_avg_ms': 19.6, 'dns_query_max_ms': 79, 'error': False}}
Consumption UPDATE: 0 of 0 bytes (0.0B).
```

marcwitasee, 2022-08-05:
Fix attempt: https://github.com/chicago-cdac/nm-exp-active-netrics/commit/e1798d7ba2c291c3c0cbfba392fe3044cf16b793

@feamster @ggmartins

marcwitasee, 2022-08-05:
Deployed fix on my beta device at home: `980b` for testing.

@feamster @ggmartins

marcwitasee, 2022-08-08:
Everything looks stable on my device after deploying the changes in https://github.com/chicago-cdac/nm-exp-active-netrics/commit/e1798d7ba2c291c3c0cbfba392fe3044cf16b793 @ggmartins @feamster. See: https://wilson.cs.uchicago.edu:8088/d/eazT3kunz/20210518-c22d980b-everywhere-wireless?orgId=1&from=now-7d&to=now&viewPanel=11. Interestingly, it looks like the mean DNS response time has actually increased after filtering out the tests to IP addresses.

marcwitasee, 2022-08-08:
Deploying to Tigerteam devices. Will deploy to Wilson devices after testing for a few days.

closed 2022-08-08 by marcwitasee