internet-innovation/nm-exp-active-netrics #113: bug: new ndt-client json output

ggmartins, 2023-02-13:
old:
```
{
	"ServerFQDN": "ndt-mlab3-ord03.mlab-oti.measurement-lab.org",
	"ServerIP": "[REDACTED]",
	"ClientIP": "[REDACTED]",
	"DownloadUUID": "ndt-j6ld9_1673569719_00000000002493B8",
	"Download": {
		"Value": 204.9798363221392,
		"Unit": "Mbit/s"
	},
	"Upload": {
		"Value": 139.31888132755253,
		"Unit": "Mbit/s"
	},
```

new:
```
 {
 	"ServerFQDN": "ndt-mlab3-ord04.mlab-oti.measurement-lab.org",
 	"ServerIP": "[REDACTED]",
 	"ClientIP": "[REDACTED]",
 	"Download": {
 		"UUID": "ndt-lp5bb_1673588518_000000000020DD9D",
 		"Throughput": {
 			"Value": 905.9671880820414,
 			"Unit": "Mbit/s"
 		},
 		"Latency": {
 			"Value": 4.096,
 			"Unit": "ms"
 		},
```
I'd also add ndt_latency as part of the metrics we want to record (from Latency).

fathimamaha, 2023-02-14:
commit hash: b07bd5f8caef264c11f49d9f24272fcd5bb6dcb5

new json does not have min rtt, for now I have commented the code out:
```
{
  'ServerFQDN': 'ndt-mlab1-ord03.mlab-oti.measurement-lab.org',
  'ServerIP': '2001:5a0:4200::75',
  'ClientIP': '2601:240:cf82:2fe0::baaf',
  'Download': {
    'UUID': 'ndt-f6w78_1673784969_000000000021216B',
    'Throughput': {
      'Value': 903.2970761116662,
      'Unit': 'Mbit/s'
    },
    'Latency': {
      'Value': 4.125,
      'Unit': 'ms'
    },
    'Retransmission': {
      'Value': 0,
      'Unit': '%'
    }
  },
  'Upload': {
    'UUID': 'ndt-f6w78_1673784969_00000000001FDBCE',
    'Throughput': {
      'Value': 40.88370290283756,
      'Unit': 'Mbit/s'
    },
    'Latency': {
      'Value': 3.77,
      'Unit': 'ms'
    },
    'Retransmission': {
      'Value': 0,
      'Unit': ''
    }
  }
}
```

marcwitasee, 2023-02-28:
We need to deploy the new version with @fathimamaha. So far we have tested on three devices, but we need to deploy more broadly.

closed 2023-03-23 by marcwitasee